OnionIoT/documentation #22: Subtask #17: Compiling a package(draft)

amsiotdeveloper, 2023-11-29:
**NOTE:** _This Subtask is linked to a larger Task List_

# **Task Owner:**
Who owns this task and will approve the final deliverables? 

Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
  1. Development – the goal is to help developers learn the process for creating packages.
  2. Deployment – packages are an important part of preparing code for deployment and production 
  3. Production
           
  ### 2. Create a short statement explaining the objectives and goals for this ticket.
     
   - This chapter will focus on the process for compiling a package 
     
# **Description of the Scope:**
  ### 1. Provide context while explaining the requirement of the ticket.
  ### 2. Provide sketches and diagrams where possible.
  ### 3. Add any other context or screenshots about the feature request here.

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 
  ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.

why its helpful to have your software packaged
explain our recommended process
acknowledge there's other methods to do this, highlight why we chose this one
provides a method to easily compile (low effort, quick compile time) a package
this should be explained at a high level and then in a step-by-step way

# **Date Required By:**
  ### 1. When is the scope required to be completed?
  ### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.

amsiotdeveloper, 2023-12-12:
this ticket got pushed into development before it was approved/assigned.  please follow the process - tickets should not be moved out of the "No Status" and "Backlog" columns.  The PM is the only one who will move tickets out of those 2 columns. 

- I have updated the ticket and assigned it to @iammuze

iammuze, 2024-01-17:
Hi, @greenbreakfast modified the outlines, please review.

greenbreakfast, 2024-01-17:
I've added another round of comments to the outline cc @iammuze

iammuze, 2024-01-18:
Hi, @greenbreakfast I have attended the latest comments.

greenbreakfast, 2024-02-02:
outline approved!

iammuze, 2024-02-14:
Hi @amsiotdeveloper  this document is ready for the initial review.

amsiotdeveloper, 2024-02-16:
Good work @iammuze 
I have finished my review of this article. 
Comments included in the doc.  Let me know if you have questions.

iammuze, 2024-02-16:
Hi @amsiotdeveloper all clear on my end, no questions for now. thanks

amsiotdeveloper, 2024-02-20:
@iammuze additional comments have been added by both @greenbreakfast and I.

@greenbreakfast I think we could use a little more direction from you with regards to the comments about the docker example then @iammuze should be able to finalize this article.

greenbreakfast, 2024-02-20:
@amsiotdeveloper added my preferred approach for docker to the article

iammuze, 2024-02-21:
Hi @greenbreakfast, the comments have been resolved. I have a query within the docs.

greenbreakfast, 2024-02-22:
@iammuze responded in the document

iammuze, 2024-02-22:
@greenbreakfast  suggestions accepted. Do you think it is good to go now? Will it also be pushed under `docs/packages/`?

greenbreakfast, 2024-02-23:
@iammuze i've made a few more edits in the first 2 sections. After that, this article is good go. Please put it at `docs/packages/compile-package.md`

iammuze, 2024-02-26:
Hi @greenbreakfast created a PR please check

closed 2024-06-11 by amsiotdeveloper