OnionIoT/documentation #26: Task: Create a menu item for Terminology, Acronyms, and Definitions.

amsiotdeveloper, 2023-12-01:
# **Task Owner:**
Who owns this task and will approve the final deliverables? 

Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development: This task will help support new users as they are learning to develop with Onion products and documentation.    
        2. Deployment
        3. Production
       

  ### 2. Create a short statement explaining the objectives and goals for this ticket.

 -  The objective is to provide value to the user experience as they learn our material and give them confidence in our documentation.
 - The primary goal is to support our content writers with clear definitions of the terminology so we can provide consistency across all articles.
 - The secondary goal is to provide a reference to help our users better understand our terminology etc 

 
# **Description of the Scope:**
  ### 1. Provide context while explaining the requirement of the ticket.
- Make a suggestion where the article should go:
     - make a section in the intro or maybe create an appendix? 
- Create 2 sections in the article:
     - terminology and definitions
     - Acronyms 
 - Start with the top most ambiguous terms and define them so we can start being consistent in the new articles we are creating. 
- This is a continuous improvement task so we will add to the article as needed.
- While creating content for new articles, we will identify new ambiguous terms so we will add those words/acronyms as they come up.   
 
  ### 2. Provide sketches and diagrams where possible.
NA
  ### 3. Add any other context or screenshots about the feature request here.
NA
     
# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 
NA

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 
- We don't want to spend too much time on this task so just create the menu item and get the framework started so we can implement the continuous improves process on this article and build more content over time. 

  ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- This article will require continuous improvement.
- Let's have the top most ambiguous terms defined prior to starting new articles so we can have consistency.

  ### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- no date specified

amsiotdeveloper, 2023-12-06:
This can become the "Glossary" section in the menu so I think let's put this ticket on hold for now like @mh1412 and @greenbreakfast suggested.  In future we will link all the chapters to this main glossary so we don't have to rewrite it all.

@greenbreakfast is there anything that we need to do now to our articles so we can easily link the chapter definitions into a main glossary later? 
If so, let's please make sure @iammuze is creating the new definition section correctly on these new articles so they can be easily linked in future.

I will keep this ticket "in progress" until we can clarify this point.

greenbreakfast, 2023-12-06:
Good thinking @amsiotdeveloper 

I've made a `packages` branch where we can do all of the work related to the Packages chapter. I've setup an intro page for the Packages chapter that imports content from another markdown file. If we keep our glossary of terms in separate markdown files, we can later pull that into a large glossary.

So https://github.com/OnionIoT/documentation/blob/packages/docs/packages/intro.md is the intro page.
And https://github.com/OnionIoT/documentation/blob/packages/docs/packages/_definitions_packages.mdx is the file that should hold the glossary of terms.

Right now the Packages chapter has one of the articles from another chapter as a placeholder - the system doesn't allow for empty categories. We'll get rid of that placeholder article when the first article for this chapter is ready.

One wrinkle here: the automatically generated table of contents does not contain the Markdown headings from imported files - this is a technical limitation on the Docusaurus side. We'll keep an eye on their development, it sounds like there's a solution on the horizon.

closed 2024-06-11 by amsiotdeveloper