OnionIoT/documentation #29: Task: OPKG Package Manager

amsiotdeveloper, 2023-12-07:
# **Task Owner:**
Who owns this task and will approve the final deliverables? 

Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development - having a clear understanding of Opkg is very important for new developers
        2. Deployment - Opkg are instrumental in the management and deployment of firmware
        3. Production
           
  ### 2. Create a short statement explaining the objectives and goals for this ticket.
     
# **Description of the Scope:**
  ### 1. Provide context while explaining the requirement of the ticket.

https://docs.google.com/document/d/1REkdylRPVTmc20dzoUQLXTIpXiAtY8GOthdKFCWFxEM/edit?pli=1#heading=h.7az4cc7gi6kc

  ### 2. Provide sketches and diagrams where possible.
  ### 3. Add any other context or screenshots about the feature request here.

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 

- create an article approximately 500 words long 

  ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.

- after reading the article, a new user should understand the importance of packages and how they are used to build an application

# **Date Required By:**
  ### 1. When is the scope required to be completed?
no specified date

  ### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
no dependencies

amsiotdeveloper, 2023-12-14:
More revisions required as per comments in the googledoc

amsiotdeveloper, 2023-12-19:
Great job finding the 3rd person professional business tone. 
Doc reviewed, a few small grammatical suggestions.
Style guide consistency needs work.

greenbreakfast, 2024-01-09:
@iammuze I've added some suggestions to the draft - it'll be ready to go after that.

Please add this article to the `packages` branch and make PR to the `main` branch.
You can put the file in the `docs/packages` directory.
And the article should be added to the `Packages` chapter in the sidebar.

iammuze, 2024-01-10:
> @iammuze I've added some suggestions to the draft - it'll be ready to go after that.
> 
> Please add this article to the `packages` branch and make PR to the `main` branch. You can put the file in the `docs/packages` directory. And the article should be added to the `Packages` chapter in the sidebar.

Clear @greenbreakfast

closed 2024-06-11 by amsiotdeveloper