OnionIoT/documentation #31: Task: Create a Custom Package

amsiotdeveloper, 2023-12-11:
# **Task Owner:**
Who owns this task and will approve the final deliverables? 

- Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development - having a clear understanding of Opkg is very important for new developers
        2. Deployment - Opkg are instrumental in the management and deployment of firmware
        3. Production
           
  ### 2. Create a short statement explaining the objectives and goals for this ticket.
     
# **Description of the Scope:**
  ### 1. Provide context while explaining the requirement of the ticket.

- The goal is to make it easy for users to create custom packages.

  ### 2. Provide sketches and diagrams where possible.
  ### 3. Add any other context or screenshots about the feature request here.

https://docs.google.com/document/d/1WgRpjrpaq_3ylE1A-xweDexls2jhbJabQNkmCvFiM34/edit?pli=1

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 

- Provide an article that makes it easy for users to understand how to create a custom package.
- provide a clear succinct process 

  ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.

- approximately 500 word article 

# **Date Required By:**
  ### 1. When is the scope required to be completed?

- No specified date

  ### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.

- No dependencies

amsiotdeveloper, 2023-12-14:
@iammuze and @greenbreakfast will review/revise the outline.
After the outline has been agreed, then @iammuze will continue with the content.

greenbreakfast, 2024-01-16:
@iammuze added comments to outline. please review

iammuze, 2024-01-17:
@greenbreakfast modified the outlines, please review.

greenbreakfast, 2024-01-17:
I've added another round of comments to the outline cc @iammuze

iammuze, 2024-01-18:
Hi, @greenbreakfast I have attended the latest comments.

greenbreakfast, 2024-01-19:
@iammuze I've added a few more comments. Once you address them, feel free to proceed with the article

iammuze, 2024-02-06:
Hi @amsiotdeveloper , this doc is ready for initial review.

amsiotdeveloper, 2024-02-09:
@iammuze I have reviewed the article.  Suggestions and comments have been added to the doc for your revision.

iammuze, 2024-02-10:
@amsiotdeveloper comments and suggestions have been addressed. 

Hi @greenbreakfast now, you can review it.

greenbreakfast, 2024-02-12:
@iammuze i've reviewed the draft and added comments

@amsiotdeveloper I agree that we should be clear on the method we recommend to build packages. This article is meant to introduce the concept and structure of a custom package. I think a separate article - on compiling packages perhaps - should introduce the sdk/build system etc. 
I've added this in comments in the draft.

iammuze, 2024-02-13:
Hi, @greenbreakfast comments have been addressed.

amsiotdeveloper, 2024-02-13:
@greenbreakfast are you suggesting we add the build system introduction to the compiling packages article here in ticket 22? 

https://docs.google.com/document/d/1Wg-UE7-tAQxdZwGA5cr8Ars8Qkv3IbFtr99wRwGap9g/edit

My suggestion is to introduced the recommended solution for the build system, then introduce a new chapter later(if needed) to address other build system options/configurations. But let's stay focused on the recommended solution.

iammuze, 2024-02-14:
Hi @greenbreakfast I have updated the `create a custom package` section with the given pointers.

iammuze, 2024-02-14:
Hi @amsiotdeveloper, please use the updated link for the `Compile A Package` chapter: https://docs.google.com/document/d/1C_scQXgbSaBRwDzTBTTTT9AG9SyoByXNtkvnIhw1O-U/edit?usp=sharing.

greenbreakfast, 2024-02-14:
Hi @iammuze, see updates in the doc

@amsiotdeveloper re: 
> are you suggesting we add the build system introduction to the compiling packages article here in ticket 22?

Yes, we've reworked the article outline and it includes a brief introduction about the SDK.
See this section in the outline https://docs.google.com/document/d/1I4uVpQeeZqDSkYZ6AsAw5vFScbymLYmL8slFtXikJ7E/edit#heading=h.7w06154mv3nf

iammuze, 2024-02-15:
Hi, @greenbreakfast the comments have been addressed.

greenbreakfast, 2024-02-16:
@iammuze I've responded to the open comments and asked @amsiotdeveloper to weigh in as well.
Once that last item is settled, we'll be ready for a PR

amsiotdeveloper, 2024-02-20:
@iammuze I agree with @greenbreakfast comments.  Please clean up the bottom of the article as per the comments in the article.

iammuze, 2024-02-21:
@greenbreakfast @amsiotdeveloper, I've addressed the latest comment in the article section and converted it into Markdown format. Is it ready to be pushed? Please let me know the exact location within the packages directory where it should be pushed.

greenbreakfast, 2024-02-22:
@iammuze please put it at `docs/packages/custom-package.md`

closed 2024-02-23 by greenbreakfast