OnionIoT/documentation #36: 1-Wire Moved to GPIO2 - update article

amsiotdeveloper, 2024-01-24:
# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development - explain the updates for the 1-wire interface 
        2. Deployment
        3. Production
           
  ### 2. Create a short statement explaining the objectives and goals for this ticket.
1-wire support has been moved from GPIO19 to GPIO2 because there was a conflict on GPIO19 if the application requires the PWM channel on GPIO19.       

# **Description of the Scope:**
  ### 1. Provide context while explaining the requirement of the ticket.

- update the 1-wire article accordingly to demonstrate how to use GPIO2 
- also please make the following edits:
•	Enable One-Wire Device Support section: need to add a step before step 1 to do opkg update
•	Find Connected One-Wire Devices section, Step 4: need to add space in command after cat

  ### 2. Provide sketches and diagrams where possible.
  ### 3. Add any other context or screenshots about the feature request here.

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 

original ticket [https://github.com/OnionIoT/OpenWRT-Packages/issues/53](url)
1-wire DT ticket [https://github.com/orgs/OnionIoT/projects/1/views/1?pane=issue&itemId=47378582](url)

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 

- the article should clearly explain how to use GPIO2 for 1-wire applications and how to setup opkg accordingly 
  ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.

- revised 1 wire article 

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no specified date

  ### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- no dependencies

amsiotdeveloper, 2024-01-29:
A couple ideas we need to consider for our revisions:
- The goal with our device tree overlay packages (one-wire, software spi) is to show what's possible and provide an example customers can work with, not necessarily to set hard guidelines.
- We should make it clear in our one-wire documentation article that this is based on a device tree overlay and the gpio is changeable.

iammuze, 2024-03-12:
@amsiotdeveloper @greenbreakfast, please share your feedback on `1-Wire Moved to GPIO2`. I have created a PR based on the new updates.

amsiotdeveloper, 2024-03-16:
@iammuze a couple things still needed to finish this ticket

iammuze, 2024-03-18:
@amsiotdeveloper  suggestions have been accepted. Additionally, this PR contains the updated changes mentioned in this ticket, so you can also review them here.