OnionIoT/documentation #44: Story: Hardware Chapter

amsiotdeveloper, 2024-02-22:
# **The Story for a New Chapter:**
### **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

### **The Scope:**
###  1. How does this task relate to the value track(s)?
- All of the effort we put into improving our product lines should provide value to our customers.
- We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
1. Development
2. Deployment
3. Production
 This chapter will touch on all 3 value tracks.
        
###  2. Create a short statement explaining the objectives and goals for this Chapter.
The objectives of this chapter are to create articles detailing the hardware. 
- The goal is to utilize as much existing content as possible so we can progress quickly through these articles. 
- Articles should be succinct, and minimalist.  
- These are MVP articles, and we will expand on them as users provide feedback and requests.  

### 3. **Reference Material:**
- Provide hyperlinks for helpful and related information. 
https://docs.google.com/spreadsheets/d/1ZLPo1iLGbPXOLvLhcrrmB9wG07G94j2TU6RlS2a7ECU/edit#gid=0

## **Deliverables or Desired Outcome:**
###  1. What are the task owners expectations? 
- After reading these articles the users should be able to evaluate hardware for new designs.  
- They should be able to find the information they need to support their hardware design efforts.
- These articles should become a good reference tool for developers. 

###  2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- MVP: Minimal viable product.  Keep it short, and let's move through these articles quickly.
- Utilize existing content where possible.
- Getting the MVP articles online quickly is important for this chapter and we have time to make improvements later. 

# **Date Required By:**
###  1. When is the scope required to be completed?
no specified date

###  2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
no dependencies 

# **The Workflow:**
- [ ] Subtask 1 (Workflow reference 10): Subject expert created a document of ideas for the new chapter.
- [ ] SME approved the chapter outline.

Subtask 3: PM will create List of tickets for the articles to be created in this chapter:
(Workflow reference 12)
- [ ] Task 1 for Article 5.1: Pinouts 
- [ ] Task 2 for Article 5.2: GPIO's
- [ ] Task 3 for Article 5.3: Pin Multiplexing 
- [ ] Task 4 for Article 5.4: Special Pins
- [ ] Task 5 for Article 5.5: Peripherals
- [ ] Task 6 for Article 5.5.1: UART
- [ ] Task 7 for Article 5.5.2: SPI
- [ ] Task 8 for Article 5.5.3: I2C
- [ ] Task 9 for Article 5.5.4: One-Wire
- [ ] Task 10 for Article 5.5.5: Hardware PWM
- [ ] Task 11 for Article 5.5.6: USB
- [ ] Task 12 for Article 5.5.7: SDIO/eMMC
- [ ] Task 13 for Article 5.5.8: Ethernet
- [ ] Task 14 for Article 5.5.9: WiFi Antenna
- [ ] Task 15 for Article 5.6: Device Tree Overlay
- [ ] Task 16 for Article 5.6.1: Device Tree Overview
- [ ] Task 17 for Article 5.6.2: Software SPI Bus
- [ ] Task 18 for Article 5.6.3: One-Wire Devices
- [ ] Task 19 for Article 5.7: Hardware Accessories
- [ ] Task 20 for Article 5.7.1: LED Chains
- [ ] Task 21 for Article 5.7.2: LTE Modem

amsiotdeveloper, 2024-02-22:
@greenbreakfast please review the hardware chapter outline.  With your approval, I will start making tickets for the list of articles. 
The objective is to have a smooth transition from from OpenWRT chapter to the hardware chapter.   The goal is to have a few article outlines ready for @iammuze by the time the OpenWRT chapter is done.

greenbreakfast, 2024-05-28:
Just need to complete the last 3 tasks:
- [ ] make sure article listing in sidebar matches ordering from this ticket
- [ ] make sure all articles have comment sections enabled
- [ ] remove deprecated articles

amsiotdeveloper, 2024-06-05:
the last 3 tasks have been completed and live

greenbreakfast, 2024-06-05:
great work guys!

closed 2024-06-05 by greenbreakfast