OnionIoT/documentation #47: Migration to Major Release Part 2

amsiotdeveloper, 2024-02-23:
### **Task Owner:**
Who owns this task and will approve the final deliverables? 

Lazar

### **The Scope:**
###  1. How does this task relate to the value track(s)?
- All of the effort we put into improving our product lines should provide value to our customers.
- We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:

- This scope is working with the new OpenWRT release so it has a major impact on all 3 value tracks
1. Development
2. Deployment
3. Production
            
###  2. Create a short statement explaining the objectives and goals for this Chapter.

- The goal is to release firmware based on the new OpenWRT release built with Onion’s customized SDK and Image Builder 

> In Part 1 Ticket #45 , we resolved any migration specific issues. Now that we have a running firmware on the new release, let’s add in Onion’s customizations

### 3. **Reference Material:**
- Provide hyperlinks for helpful and related information. 

- OnionIoT/openwrt-buildsystem-wrapper

## **Deliverables or Desired Outcome:**
###  1. What are the task owners expectations? 

Steps

1. Create a new branch in the openwrt-buildsystem-wrapper repo
    1. The branch should be based on the previous release and should be named to match OpenWRT release name, for example `openwrt-22.03` for OpenWRT 22.03
2. Update the openwrt-buildsystem-wrapper repo configuration to use the build system commit used for the target release
3. Successfully compile the build system with our customizations
    - Debug compilation issues as needed by updating patches and our config files
    - To move forward:
        - build system needs to compile without errors
        - output SDK should be tested
        - output Image Builder should be tested
4. Build custom packages and new firmware based on the customized SDK and Image Builder
    1. Update the openwrt-imagebuilder-wrapper repo configuration to use **Onion’s customized** Image Builder for the target release
    2. Update the openwrt-sdk-wrapper repo configuration to use  **Onion’s customized** SDK for the target release
    3. Build packages and fimrware

###  2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.

1. Implement the mentioned steps.
2. Identify and address any issues that arise.
3. Implement the tests in the checklist below and provide test results accordingly. 

# **Date Required By:**
###  1. When is the scope required to be completed?
- no specified completion date
###  2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- Part 1 Ticket #45 and Ticket #46 must be completed before starting this ticket. 

# **The Workflow:**
- [ ] Subtask 1: Execute the mentioned steps to perform Part 2 of the migration.
- [ ] Subtask 2: Execute the test process in Ticket #48 
- [ ] Subtask 3: Review test results for approval to release the new firmware.

greenbreakfast, 2024-02-23:
@amsiotdeveloper 

> 4. Build custom packages and new firmware based on the customized SDK and Image Builder
>
>    1. Update the openwrt-imagebuilder-wrapper repo configuration to use **Onion’s customized** Image [...]

building off [my comment](https://github.com/OnionIoT/documentation/issues/45#issuecomment-1961558933) in #45:

The updates to the openwrt-imagebuilder-wrapper and openwrt-sdk-wrapper configuration should take place in the `imagebuilder-profile` and `sdk-profile` files in the OnionIoT/openwrt-packages repo

greenbreakfast, 2024-05-08:
closing this issue
replaced with https://github.com/OnionIoT/openwrt-buildsystem-wrapper/issues/24

closed 2024-05-08 by greenbreakfast