OnionIoT/documentation #51: Article 5.1: Pinouts

amsiotdeveloper, 2024-02-29:
**NOTE:** _This Subtask is linked to a larger Task List_ticket #44 

# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development - The hardware chapter is important for evaluating the Omega modules for a new project and supporting developers in the development process.
        2. Deployment
        3. Production
           
  ### 2. Create a short statement explaining the objectives and goals for this ticket.
- This article will provide an outline of the pins for each module.

- The goal is to utilize as much existing content as possible so we can progress quickly through these articles.
     
# **Description of the Scope:**
  ### 1. Provide context while explaining the requirement of the ticket.
  ### 2. Provide sketches and diagrams where possible.
  ### 3. Add any other context or screenshots about the feature request here.

Pinouts
Display in a toggle like in the Pin Multiplexing article (https://github.com/OnionIoT/documentation/blob/main/docs/hardware-interfaces/pin-multiplexing.md?plain=1#L18) 
Omega2 pinout: use https://github.com/OnionIoT/Onion-Media/blob/master/Pinouts/Omega2.png
Omega2S pinout: use https://github.com/OnionIoT/Onion-Media/blob/master/Pinouts/Omega2S.png
List of interfaces
Have a listing of supported interfaces for the Omega2/2+ vs Omega2S/Omega2S+
Can be based on lists from https://docs.onion.io/omega2-docs/omega2p.html#the-omega2-at-a-glance-1 and https://docs.onion.io/omega2-docs/omega2s.html#variants
But only include the hardware interfaces, not the Processor, memory, etc

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 

- To practically illustrate this, I've updated the I2C article based on the contents from the outline. It's a minimum viable article and it can be used as a template for the rest of the articles in this chapter.
Find it available here: https://github.com/OnionIoT/documentation/blob/hardware/docs/hardware-interfaces/i2c.md

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 

- Articles should be succinct, and minimalist.
- These are MVP articles, and we will expand on them as users provide feedback and requests.
- All articles from the Hardware chapter will go in the hardware branch. 

  ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- an article that follows our current style guide

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no date has been specified 

### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- Ticket #44 has a dependency on this ticket to be completed

iammuze, 2024-03-12:
Hi! @amsiotdeveloper @greenbreakfast  comments have been addressed.

amsiotdeveloper, 2024-03-15:
@iammuze I have reviewed finished reviewing the article.  Basically the hardware articles are much shorter and we will keep them minimal so I think the diagrams and table says pretty much all that needs to be said in this article. 
When publishing the final article, let's make sure the diagrams are large enough so the pin descriptions are easily read.  They look low res and a little pixelated in the goolge doc so increasing the size will help improve readability in the final article.

iammuze, 2024-03-18:
Alright @amsiotdeveloper , I've just slightly reduced the size in the Google Doc, but I'll make sure to use the original res for the final article. Is this article ready for markdown and PR?

greenbreakfast, 2024-03-20:
Review complete, added comments to article draft

iammuze, 2024-03-21:
@greenbreakfast comments and suggestions have been addressed.

greenbreakfast, 2024-03-25:
article is ready for production

iammuze, 2024-03-27:
@greenbreakfast PR is created and ready for your review.

closed 2024-06-11 by amsiotdeveloper