OnionIoT/documentation #52: Article 5.1: GPIO

amsiotdeveloper, 2024-02-29:
**NOTE:** _This Subtask is linked to a larger Task List_ticket #44 

# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development - The hardware chapter is important for evaluating the Omega modules for a new project and supporting developers in the development process.
        2. Deployment
        3. Production
           
  ### 2. Create a short statement explaining the objectives and goals for this ticket.

**Intro**
- GPIOs (General Purpose Input Outputs) controllable by the user
- Can be used for interacting with external circuits

**Hardware**
- Table listing number of GPIOs on Omega2/2+ vs Omega2S/Omega2S+
- Electrical Ratings
     - Reuse content from https://docs.onion.io/omega2-docs/using-gpios.html#gpio-electrical-ratings
- Multiplexed Pins
     - A number of the available pins can be used for multiple purposes other than general purpose input/output when needed. Such pins are referred to as multiplexed pins.
     - Point to 5.3: Pin Multiplexing article

**Software**
- Intro
     - The GPIOs can be accessed through the GPIO sysfs interface, see the [documentation](https://www.kernel.org/doc/Documentation/gpio/sysfs.txt) for details.
     - The GPIO sysfs interface is deprecated but is currently the best option for userspace GPIO access. More context available in [this post by Luz on the Onion Community](https://community.onion.io/topic/4892/can-bus-using-mcp2515-with-omega2/13).
- Note on GPIO numbering
     - An important note: because of changes in the kernel, the GPIOs are not numbered like they were before:
          - GPIO 0 - 31 ⇒ GPIO 480 - 511 (GPIO n + 480)
          - GPIO 32 - 63 ⇒ GPIO 448 - 479 (GPIO n + 416)
     - Include section on gpio-looup utility
- Interacting with GPIOs
     - Command line
          - Use the GPIO sysfs interface
          - Good for basic testing
          - Usage instructions: link to https://www.kernel.org/doc/Documentation/gpio/sysfs.txt 
     - C/C++ 
          - Recommended library: libi2c
          - How to use the library: See source code at https://git.kernel.org/pub/scm/utils/i2c-tools/i2c-tools.git/tree/
          - Future notes: look at
               - https://github.com/amaork/libi2c
     - Python
               - Recommended module: python3-smbus
               - Installation instructions: 
                    - Opkg update

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 

- To practically illustrate this, I've updated the I2C article based on the contents from the outline. It's a minimum viable article and it can be used as a template for the rest of the articles in this chapter.
Find it available here: https://github.com/OnionIoT/documentation/blob/hardware/docs/hardware-interfaces/i2c.md

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 

- Articles should be succinct, and minimalist.
- These are MVP articles, and we will expand on them as users provide feedback and requests.
- All articles from the Hardware chapter will go in the hardware branch. 

  ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- a succinct article that follows our current style guide based on the original article

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no date has been specified 

### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- Ticket #44 has a dependency on this ticket to be completed

amsiotdeveloper, 2024-03-16:
@iammuze I have completed my review.  Please see revisions and notes.

iammuze, 2024-03-18:
Hi, @amsiotdeveloper, comments and suggestions have been addressed.

greenbreakfast, 2024-03-20:
Review complete, added comments
@iammuze with the articles in this hardware chapter, we can reuse lots of the content that's already in the article (in this case in https://github.com/OnionIoT/documentation/blob/main/docs/hardware-interfaces/gpio.md). 
This is mentioned in the outline for this article - there was no need to write a new section for the gpio-lookup utility.

iammuze, 2024-03-21:
Alright @greenbreakfast, I have updated the `gpio-lookup utility` section based on the provided article link. Comments and suggestions have been addressed.

greenbreakfast, 2024-03-25:
article approved for production

iammuze, 2024-03-27:
@greenbreakfast added 'GPIO' article for review.

closed 2024-06-11 by amsiotdeveloper