OnionIoT/documentation #53: Article 5.3: Pin Multiplexing

amsiotdeveloper, 2024-02-29:
**NOTE:** _This Subtask is linked to a larger Task List_ticket #44 

# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development - The hardware chapter is important for evaluating the Omega modules for a new project and supporting developers in the development process.
        2. Deployment
        3. Production
           
  ### 2. Create a short statement explaining the objectives and goals for this ticket.

- Rework the existing article to be separated into intro, hardware, software sections

- Intro
     - Keep current intro
     
- Hardware
     - Keep pinout
     
- Software
     - Intro
          - The omega2-ctrl utility is a user-friendly way to change the pin functionality
               - The pin multiplexing configuration is controlled by a programmable register of the Omega’s SoC
               - The Onion Omega2 firmware enables the /dev/mem device that is used to modify physical memory
               - The omega2-ctrl program safely modifies the registers that configures pin multiplexing
     - View Current Multiplexing Configuration
               - Keep current content
     - Changing Multiplexing Configuration - Change pin function
               - Keep current content

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 

- To practically illustrate this, I've updated the I2C article based on the contents from the outline. It's a minimum viable article and it can be used as a template for the rest of the articles in this chapter.
Find it available here: https://github.com/OnionIoT/documentation/blob/hardware/docs/hardware-interfaces/i2c.md

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 

- Articles should be succinct, and minimalist.
- These are MVP articles, and we will expand on them as users provide feedback and requests.
- All articles from the Hardware chapter will go in the hardware branch. 

  ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- a succinct article that follows our current style guide based on the original article

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no date has been specified 

### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- Ticket #44 has a dependency on this ticket to be completed

amsiotdeveloper, 2024-03-26:
@iammuze everything looks good on this article.  Just a couple small things.
Please update and move the ticket to in review when changes are done.

amsiotdeveloper, 2024-03-26:
@greenbreakfast this ticket is ready for your review/approval

greenbreakfast, 2024-03-28:
good to go, please create a PR

greenbreakfast, 2024-04-03:
merged PR on April 1st

closed 2024-04-03 by greenbreakfast