OnionIoT/documentation #54: Article 5.4 Special Pins

amsiotdeveloper, 2024-02-29:
**NOTE:** _This Subtask is linked to a larger Task List_ticket #44 

# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development - The hardware chapter is important for evaluating the Omega modules for a new project and supporting developers in the development process.
        2. Deployment
        3. Production
           
  ### 2. Create a short statement explaining the objectives and goals for this ticket.

Can largely reuse content from https://docs.onion.io/omega2-docs/using-gpios.html#important-special-gpios

- Intro
     - Information on pins with unique properties
- System boot pins
     - Reuse content from https://docs.onion.io/omega2-docs/using-gpios.html#boot-pins
- SPI Pins
     - Reuse content from https://docs.onion.io/omega2-docs/using-gpios.html#spi-pins
     - Also see section 3.3.2 in the Omega2S datasheet, reuse any content that’s relevant: 
     - https://github.com/OnionIoT/Omega2/blob/master/Documents/Omega2S%20Datasheet.pdf
- Reset Pins
     - Reuse content from https://docs.onion.io/omega2-docs/using-gpios.html#reset-led-pins
     - Additional useful context that should be included from section 3.3.3  Omega2S datasheet: 
     -  https://github.com/OnionIoT/Omega2/blob/master/Documents/Omega2S%20Datasheet.pdf
- LED Pins
     - Reuse content from https://docs.onion.io/omega2-docs/using-gpios.html#reset-led-pins
     - Note: difference between Omega2/2+ vs Omega2S/2S+ 
          - on the Omega2/Omega2+ the pin is not exposed, but directly connected to an LED
          - On the Omega2S/2S+ the pin is exposed
- Power Supply pins
     - Omega2/2+
          - 3.3V Vin
          - GND
     - Omega2S/2S+
          - 2x 3.3V
          - VDD_FLASH
          - Reference section 3.3.4 in  Omega2S datasheet: 
          -  https://github.com/OnionIoT/Omega2/blob/master/Documents/Omega2S%20Datasheet.pdf
- Other
     - See https://docs.onion.io/omega2-docs/using-gpios.html#pins-at-boot
     - GPIO11 - Will settle at Digital High - Used to provide power for reset button on Omega2 Docks
     - The behavior of these pins during boot is goverened by the Omega’s bootloader.

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 

- To practically illustrate this, I've updated the I2C article based on the contents from the outline. It's a minimum viable article and it can be used as a template for the rest of the articles in this chapter.
Find it available here: https://github.com/OnionIoT/documentation/blob/hardware/docs/hardware-interfaces/i2c.md

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 

- Articles should be succinct, and minimalist.
- These are MVP articles, and we will expand on them as users provide feedback and requests.
- All articles from the Hardware chapter will go in the hardware branch. 

  ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- a succinct article that follows our current style guide based on the original article

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no date has been specified 

### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- Ticket #44 has a dependency on this ticket to be completed

iammuze, 2024-03-21:
@amsiotdeveloper  comments and suggestions have been addressed.

greenbreakfast, 2024-03-28:
Added comments after review @iammuze

iammuze, 2024-03-29:
@greenbreakfast  draft is ready for your review again

greenbreakfast, 2024-04-03:
Added one more round of comments/suggestions @iammuze , will be ready for PR after those are handled