OnionIoT/documentation #56: Article 5.5.2 SPI

amsiotdeveloper, 2024-03-04:
**NOTE:** _This Subtask is linked to a larger Task List_ticket #44 

# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development - The hardware chapter is important for evaluating the Omega modules for a new project and supporting developers in the development process.
        2. Deployment
        3. Production
           
  ### 2. Create a short statement explaining the objectives and goals for this ticket.

**Intro**
     - This article cover the SPI port

**Context**
- The Serial Peripheral Interface (SPI) is a four-wire synchronous communication protocol, largely used to connect microprocessors or microcontrollers to sensors, memory, and other peripherals.
SPI is a synchronous data bus, meaning that one of the lines is a clock, used to synchronize the bits being sent on the data lines.
The protocol is based on the Master-Worker architecture. 
     - The Master will generate the System Clock and the Chip Select signals
     - Each worker requires its own Chip Select connection to the Master.

**Hardware**
- One SPI interface is available. 
     - The interface supports half-duplex transmissions and can operate in host-mode only
     - The maximum SPI clock frequency is 40 MHz
     - The flash storage occupies SPI Chip Select 0 and cannot be used, external devices can be connected to SPI Chip Select 1.
     - External devices connected to the SPI bus can affect the boot sequence under certain conditions. Refer to special pins-SPI pins for details
- Pinout diagrams of Omega2, Omega2S with SPI pins and CS1 highlighted

**Software**
- SPI interface available at /dev/spidev0.1
     - spidev0 is the Omega2’s SPI bus number.
     - 1 indicates the device ID. This corresponds to the worker connected to the Omega2’s CS1 pin.
- Interacting with the SPI bus
     - Python
          - Recommended module: python3-spidev
          - For installation and usage instructions see: https://github.com/OnionIoT/python-spidev#python-spidev

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 

- Refer to the original article

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 

- Articles should be succinct, and minimalist.
- These are MVP articles, and we will expand on them as users provide feedback and requests.
- All articles from the Hardware chapter will go in the hardware branch. 

  ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- a succinct article that follows our current style guide based on the original article

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no date has been specified 

### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- Ticket #44 has a dependency on this ticket to be completed

remingtontr, 2024-04-12:
Ready for initial review.

amsiotdeveloper, 2024-04-16:
@remingtontr  good work!!  No suggestions for revisions from me. 

@greenbreakfast this article is ready for your final review.

This article is very basic, so I think in future after the MVP is done we might want to include some of the more advanced details of the SPI bus like clock polarity, idle hi/low etc. This is certainly a more advanced topic but necessary when trying to interface with various types of SPI devices.  This is good enough for now though.

greenbreakfast, 2024-04-22:
@remingtontr added 2 comments, otherwise looks good

greenbreakfast, 2024-04-25:
@remingtontr thanks for addressing the comments + suggestions, this is ready for a PR