OnionIoT/documentation #58: 5.5.4 One-Wire

amsiotdeveloper, 2024-03-04:
**NOTE:** _This Subtask is linked to a larger Task List_ticket #44 

# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development - The hardware chapter is important for evaluating the Omega modules for a new project and supporting developers in the development process.
        2. Deployment
        3. Production
           
  ### 2. Create a short statement explaining the objectives and goals for this ticket.

**Intro**
- Reuse intro from existing one-wire document: https://documentation.onioniot.com/device-tree-overlay/one-wire-devices/

**Context**
- What is a one-wire device
     - Reuse existing content: https://documentation.onioniot.com/device-tree-overlay/one-wire-devices/#what-is-a-one-wire-device
- What is the one-wire protocol
     - Reuse existing content https://documentation.onioniot.com/device-tree-overlay/one-wire-devices/#what-is-a-one-wire-protocol

**Hardware**
- Support for one-wire devices on the Omega2 is enabled with a Device Tree Overlay - a run-time modification to the kernel’s live tree
     - In the example device tree overlay package for one-wire support provided by Onion, the one-wire bus master is set to GPIO2
          - This can be changed by creating a custom version of the one-wire device tree package 
     - See the Device tree overlay chapter for details - link to 5.6 device tree overlay chapter
- Connect a One-Wire Device to the Omega2
     - Reuse existing content https://documentation.onioniot.com/device-tree-overlay/one-wire-devices/#connect-a-one-wire-device-to-the-omega2

**Software**
- Enable One-Wire Device Support
- Find Connected One-Wire Devices
- Read Data from a Connected One-Wire Device
- Common Pitfalls and Precautions
- Reuse existing content for the above

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 

- Articles should be succinct, and minimalist.
- These are MVP articles, and we will expand on them as users provide feedback and requests.
- All articles from the Hardware chapter will go in the hardware branch. 

  ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- a succinct article that follows our current style guide based on the original article

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no date has been specified 

### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- Ticket #44 has a dependency on this ticket to be completed

amsiotdeveloper, 2024-04-16:
@remingtontr I like the consistency I'm seeing from document to document!  Well done. Just a few suggestions for you to review in this article please.

greenbreakfast, 2024-04-22:
looks good @remingtontr 
For this PR, please make a new file under `docs/hardware-interfaces` and add it to the hardware interfaces chapter in sidebars.js

The plan is to change the one-wire article in the device tree overlay chapter to talk more about the device-tree overlay specific parts of one-wire (section 5.6.3 in hw chapter outline)

closed 2024-06-11 by amsiotdeveloper