OnionIoT/documentation #59: Article 5.5.5 Hardware PWM

amsiotdeveloper, 2024-03-04:
**NOTE:** _This Subtask is linked to a larger Task List_ticket #44 

# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development - The hardware chapter is important for evaluating the Omega modules for a new project and supporting developers in the development process.
        2. Deployment
        3. Production
           
  ### 2. Create a short statement explaining the objectives and goals for this ticket.

**Intro**
- The Omega’s SoC has a number of PWM modules. The Omega2 exposes two (2) PWM hardware modules on its pin headers while the Omega2S exposes all four (4).

**Context**
- PWM (Pulse-width modulation) is a technique of producing varying analog signals from a digital source
     - Duty cycle = percentage of time the signal is high (0-100)
     - Frequency = signal frequency

**Hardware**
- Hardware-based PWM
     - Omega2/2+: 2 channels, Omega2S/2s+: 4 channels
     - The hardware PWM modules have a frequency range of 39Hz to 40MHz. 
          - For lower frequencies it is recommended to consider an external hardware PWM generator or the use of a software generated PWM signal from a GPIO pin.
     - Include a table of the pins - see https://documentation.onioniot.com/hardware-interfaces/pwm#enabling-pwm-pins
- Pinout diagrams of Omega2, Omega2S with PWM pins highlighted

**Software**
- Enabling Hardware PWM
     - Reuse existing section in document - https://documentation.onioniot.com/hardware-interfaces/pwm#enabling-hardware-pwm
- Adjusting Pin Multiplexing to Enable PWM Pins
     - Reuse existing section in document - https://documentation.onioniot.com/hardware-interfaces/pwm#enabling-pwm-pins
- Generating PWM signals
     - Reuse existing section in document - https://documentation.onioniot.com/hardware-interfaces/pwm#generating-pwm-signals
     - Include a note about the script:
          - Helpful wrapper around sysfs for pwm
          - Source is available at: https://github.com/OnionIoT/OpenWRT-Packages/blob/openwrt-22.03/omega2-base/files/usr/lib/onion-pwm-lib.sh 
- Note: can remove the “Writing your own script to generate PWM signals” section

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 

- Articles should be succinct, and minimalist.
- These are MVP articles, and we will expand on them as users provide feedback and requests.
- All articles from the Hardware chapter will go in the hardware branch. 

  ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- a succinct article that follows our current style guide based on the original article

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no date has been specified 

### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- Ticket #44 has a dependency on this ticket to be completed

remingtontr, 2024-04-15:
Ready for initial review.

greenbreakfast, 2024-04-25:
@remingtontr looks good with the updates, ready for a PR