OnionIoT/documentation #61: Article 5.5.7 SDIO/eMMC

amsiotdeveloper, 2024-03-04:
**NOTE:** _This Subtask is linked to a larger Task List_ticket #44 

# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development - The hardware chapter is important for evaluating the Omega modules for a new project and supporting developers in the development process.
        2. Deployment
        3. Production
           
  ### 2. Create a short statement explaining the objectives and goals for this ticket.

**Intro**
- The Omega2 features an SDIO/eMMC interface
     - Enables use of SD/MicroSD cards or eMMC storage chips
- Omega2+ features a microSD slot
- Omega2S/2S+ expose the pins 

**Hardware**
- single SDIO interface
     - Only one of eMMC, SD/Micro-SD, or other SDIO device can be used
     - supports the SDXC specification for SD cards, with a maximum capacity of 2 TB and a maximum transfer speed of 300 MB/s, and the eMMC5.1 interface for eMMC storage.
- Photo of Omega2+ with MicroSD Slot highlighted 
     - See https://docs.onion.io/omega2-docs/using-a-microsd-card.html
- Pinout diagrams of Omega2S with SDIO/eMMC pins highlighted
     - See https://docs.onion.io/omega2-docs/using-a-microsd-card.html
- See Omega2S hardware design guide for more information and recommendations on designing boards with MicroSD card slots or eMMC: https://github.com/OnionIoT/Omega2/blob/master/Documents/Omega2S%20Hardware%20Design%20Guide.pdf

**SD Card/eMMC Storage**
- Reuse existing content: https://documentation.onioniot.com/hardware-interfaces/external-storage#sd-card

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 

- Articles should be succinct, and minimalist.
- These are MVP articles, and we will expand on them as users provide feedback and requests.
- All articles from the Hardware chapter will go in the hardware branch. 

  ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- a succinct article that follows our current style guide based on the original article

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no date has been specified 

### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- Ticket #44 has a dependency on this ticket to be completed

remingtontr, 2024-04-23:
Ready for initial review. I had to push the MicroSD image to the second page as it gets cut off if left on the first page. This will look much better in markdown.

amsiotdeveloper, 2024-04-29:
@remingtontr  this looks great.  A couple small suggestions for improvements.

amsiotdeveloper, 2024-04-30:
@greenbreakfast  this article is ready for your review.  I realize now you have intentionally used "filesystem" and it was not a typo? 
My suggestion for @remingtontr was to use "file system" which is in my opinion grammatically correct.

To clarify, if you were using a filename such as "filesystem.lib" then you could refer to it as "filesystem."
However, if you are talking about the system of files on the drive, then it is a "file system." 
This also came up in another article so I think let's make a decision and stay consistent.

greenbreakfast, 2024-05-03:
@remingtontr looks good. I added a few suggestions, please take a look
@amsiotdeveloper I agree we should be consistent - I'm good with using "file system". Can you please make a reminder ticket to add details on eMMC storage filesystems to this article?

closed 2024-05-09 by greenbreakfast