OnionIoT/documentation #62: Article 5.5.8 Ethernet

amsiotdeveloper, 2024-03-04:
**NOTE:** _This Subtask is linked to a larger Task List_ticket #44 

# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development - The hardware chapter is important for evaluating the Omega modules for a new project and supporting developers in the development process.
        2. Deployment
        3. Production
           
  ### 2. Create a short statement explaining the objectives and goals for this ticket.

**Intro**
- The Omega2 features an 10/100M Ethernet port for wired networking

**Hardware**
- Single 10/100M Ethernet integrated PHY
- Pinout diagrams of Omega2, Omega2S with Ethernet pins highlighted
- See Omega2S hardware design guide for more information and recommendations on designing boards with MicroSD card slots or eMMC: https://github.com/OnionIoT/Omega2/blob/master/Documents/Omega2S%20Hardware%20Design%20Guide.pdf

**Software**
- The ethernet port can be configured to be a client or a DHCP host. It is set to client mode by default. 
- Point to ethernet networking article: https://documentation.onioniot.com/networking/ethernet

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 

- Articles should be succinct, and minimalist.
- These are MVP articles, and we will expand on them as users provide feedback and requests.
- All articles from the Hardware chapter will go in the hardware branch. 

  ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- a succinct article that follows our current style guide based on the original article

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no date has been specified 

### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- Ticket #44 has a dependency on this ticket to be completed

remingtontr, 2024-04-24:
Ready for initial review

amsiotdeveloper, 2024-04-29:
@remingtontr  just a small oversight to fix on this one, but the rest is good to go for @greenbreakfast  to do final review/approval.

greenbreakfast, 2024-04-29:
@remingtontr added 2 comments about hyperlinks. Ready for PR after that's addressed

closed 2024-05-03 by greenbreakfast