OnionIoT/documentation #65: 5.6.1 Device Tree Overview

amsiotdeveloper, 2024-03-05:
**NOTE:** _This Subtask is linked to a larger Task List_ticket #44 

# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development - The hardware chapter is important for evaluating the Omega modules for a new project and supporting developers in the development process.
        2. Deployment
        3. Production
           
  ### 2. Create a short statement explaining the objectives and goals for this ticket.

**Intro**
- Keep existing intro content

**Context**
- Context: What are Device Trees? How do they generally work?
     - Keep existing content
- What is a device tree overlay
     - a modification to the kernel’s live tree on a device - meaning adding support for new hardware or changing how hardware is supported
- How this is implemented
     - Keep existing content
- How this works on the device
     - Install a software package (with the package manager) that with a compiled device tree overlay
     - After a reboot, the device tree overlay will be loaded and the device will have the additional 
- How to create custom Device Tree Overlay Packages
     - Keep existing content

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 

- Articles should be succinct, and minimalist.
- These are MVP articles, and we will expand on them as users provide feedback and requests.
- All articles from the Hardware chapter will go in the hardware branch. 

 ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- a succinct article that follows our current style guide based on the original article

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no date has been specified 

 ### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- Ticket #44 has a dependency on this ticket to be completed

remingtontr, 2024-05-01:
Ready for initial review.

amsiotdeveloper, 2024-05-02:
@remingtontr  this is really good. 
One item was missed from the outline so we need to work that in a little better, and I think we need to put a little more thought into one section.  Please see comments in the document.
Thank you.

remingtontr, 2024-05-02:
Gary,

I have reworked the awkward sentence and I think it now sounds better and makes more sense.

I have added in the missed sentence as a NOTE.

Tom Remington

amsiotdeveloper, 2024-05-03:
@remingtontr  looks better now.
@greenbreakfast  this article is ready for your final review/approval.  There is just one comment we need your input on so we can keep your intended meaning of the original sentence.

greenbreakfast, 2024-05-06:
@remingtontr looks good! I added one comment, after that it's good for a PR

closed 2024-05-10 by greenbreakfast