OnionIoT/documentation #66: 5.6.2 Software SPI Bus

amsiotdeveloper, 2024-03-05:
**NOTE:** _This Subtask is linked to a larger Task List_ticket #44 

# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development - The hardware chapter is important for evaluating the Omega modules for a new project and supporting developers in the development process.
        2. Deployment
        3. Production
           
  ### 2. Create a short statement explaining the objectives and goals for this ticket.

- Largely keep existing content
- In “How it’s Used” section, link to 5.5.2: SPI article for further instructions on using the SPI bus

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 

- Articles should be succinct, and minimalist.
- These are MVP articles, and we will expand on them as users provide feedback and requests.
- All articles from the Hardware chapter will go in the hardware branch. 

 ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- a succinct article that follows our current style guide based on the original article

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no date has been specified 

 ### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- Ticket #44 has a dependency on this ticket to be completed

remingtontr, 2024-04-25:
Ready for initial review

amsiotdeveloper, 2024-04-30:
@remingtontr I know the original outline was to copy original content, but I made a few suggestions for improvements. 
Please review/revise accordingly.

greenbreakfast, 2024-05-03:
@remingtontr, I added a few suggestions, please take a look

greenbreakfast, 2024-05-06:
@remingtontr, added one suggestion. Ready for PR after that.

remingtontr, 2024-05-08:
Fixed. I will convert to Markdown and push.

Tom Remington

closed 2024-05-10 by greenbreakfast