OnionIoT/documentation #67: 5.6.3 One-Wire Devices

amsiotdeveloper, 2024-03-05:
**NOTE:** _This Subtask is linked to a larger Task List_ticket #44 

# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development - The hardware chapter is important for evaluating the Omega modules for a new project and supporting developers in the development process.
        2. Deployment
        3. Production
           
  ### 2. Create a short statement explaining the objectives and goals for this ticket.

- This article will change to discuss more device-tree overlay related items

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 

**Note** please refer to the hardware article and ensure each article has unique content related to the chapter.
One article is discussing one-wire from the hardware point of view and this one is covering the device tree aspect of one-wire so there should not be any duplicate content.  
- https://github.com/OnionIoT/documentation/issues/58

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 

- Articles should be succinct, and minimalist.
- These are MVP articles, and we will expand on them as users provide feedback and requests.
- All articles from the Hardware chapter will go in the hardware branch. 

 ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- a succinct article that follows our current style guide based on the original article

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no date has been specified 

 ### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- Ticket #44 has a dependency on this ticket to be completed

greenbreakfast, 2024-04-23:
Outline updated, this is ready for development

remingtontr, 2024-05-03:
Ready for initial Review

amsiotdeveloper, 2024-05-08:
@remingtontr  looks good - just a couple small items.
@greenbreakfast  see comment - your feedback is needed please, and the article is ready for your final review.

greenbreakfast, 2024-05-08:
@amsiotdeveloper Added a comment explaining what I think we should convey in that intro section. I see how it's a little confusing currently. @remingtontr please take a look and rework as you see fit
The rest of the article looks good.

remingtontr, 2024-05-09:
Fixes completed.

Tom Remington

greenbreakfast, 2024-05-13:
@amsiotdeveloper please take a look at the latest edits and let us know if it resolves your earlier comment

greenbreakfast, 2024-05-14:
Hey @remingtontr, in this spirit of shipping fast, I think we should publish this article and add polish later if necessary.
Please go ahead with a PR

remingtontr, 2024-05-16:
Done and pushed!

Tom Remington

closed 2024-05-22 by greenbreakfast