OnionIoT/documentation #68: 5.7.1 LED Chains

amsiotdeveloper, 2024-03-05:
**NOTE:** _This Subtask is linked to a larger Task List_ticket #44 

# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development - The hardware chapter is important for evaluating the Omega modules for a new project and supporting developers in the development process.
        2. Deployment
        3. Production
           
  ### 2. Create a short statement explaining the objectives and goals for this ticket.

**Intro**
     - Keep existing content

**Hardware**
- External LED chains can be connected to any of the 4 hardware PWM channels
- Supported LED chain types:
     - WS2811
     - WS2812(B)
     - WS2813(B)
     - WS2815
     - P9823
     - SK6812
- Reuse table of PWM channels from 5.5.5 hardware pwm article

**Software**
- GPIO Mux
     - Keep existing content
     - Add link to 5.3 Pin multiplexing article for users to learn more
- Activate the LED Chain Driver
     - Keep existing content
- Update the LEDs
     - Keep existing content

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 

- Refer to the original article

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 

- Articles should be succinct, and minimalist.
- These are MVP articles, and we will expand on them as users provide feedback and requests.
- All articles from the Hardware chapter will go in the hardware branch. 

  ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- a succinct article that follows our current style guide based on the original article

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no date has been specified 

### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- Ticket #44 has a dependency on this ticket to be completed

remingtontr, 2024-05-01:
Ready for initial review.

amsiotdeveloper, 2024-05-02:
@greenbreakfast if there are 4 PWM channels, does that mean the Omega can support 4 LED strings simultaneously or just one string at a time?  I think it would be important to point out that 4 channels can be used simultaneously (assuming it is possible).
From what I have seen, there is a much larger demand for WLED with the 4 channel ESP32 boards than 1 channel.

amsiotdeveloper, 2024-05-02:
@remingtontr other than the note above about support for 4 LED strings, everything else looks good to me.  @greenbreakfast please proceed with your review and feedback on the 4 channels. 
Thank you.

greenbreakfast, 2024-05-06:
@remingtontr added one comment, good for PR after that

closed 2024-05-07 by greenbreakfast