OnionIoT/documentation #69: 5.7.2 LTE Modem

amsiotdeveloper, 2024-03-05:
**NOTE:** _This Subtask is linked to a larger Task List_ticket #44 

# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development - The hardware chapter is important for evaluating the Omega modules for a new project and supporting developers in the development process.
        2. Deployment
        3. Production
           
  ### 2. Create a short statement explaining the objectives and goals for this ticket.

**TO DO** 
**Intro**
     - This article covers the LTE Modem

**Context**

**Hardware**

**Software**
Move content from “Steps” section here

Caveats
Reuse this content

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 

- Refer to the original article

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 

- Articles should be succinct, and minimalist.
- These are MVP articles, and we will expand on them as users provide feedback and requests.
- All articles from the Hardware chapter will go in the hardware branch. 

  ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- a succinct article that follows our current style guide based on the original article

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no date has been specified 

### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- Ticket #44 has a dependency on this ticket to be completed

greenbreakfast, 2024-04-23:
Outline updated, this is ready for development

remingtontr, 2024-05-03:
Ready for initial Review. Please see the comment I've added to the document - looking for a clarification.

amsiotdeveloper, 2024-05-08:
@greenbreakfast @remingtontr the caveat section is a little confusing.  I made a couple suggestions/comments in the draft doc.

greenbreakfast, 2024-05-22:
@amsiotdeveloper @remingtontr added suggestions on how to address the last section. Let me know if this resolves the confusion

greenbreakfast, 2024-05-25:
@remingtontr I'm happy with this latest revision, please go ahead with a PR

closed 2024-05-28 by greenbreakfast