OnionIoT/documentation #82: Ethernet Networking

amsiotdeveloper, 2024-05-03:
**NOTE:** _This Subtask is linked to a larger Task List_

# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development
        2. Deployment
        3. Production
      This article is primarily targeting the Development track to help users setup and use Ethernet networking. 
           
  ### 2. Create a short statement explaining the objectives and goals for this ticket.
The objectives of this article is to provide the reader with the required information to setup and configure a Ethernet networking solution.     

# **Description of the Scope:**

For the draft scope outline please refer to:

[https://docs.google.com/document/d/1w67dYSW5Kp9E4u7liLximAxnMgJMtOyQrj-JYH162yc/edit]

Anything in _italics_ is an instruction to the writer. The rest is content

Start with the existing articles in docs/networking in the OnionIoT/documentation repo, and add to them based on the googledocs outline.

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 

See this article from old documentation: https://docs.onion.io/omega2-docs/fix-ip-addr-collisions.html

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 
- reuse content where possible
- create an article with a professional business/technical tone 
- create an article consistent with the style guide 

  ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.

working doc in google docs 
markdown pushed to repo 

# **Date Required By:**
  ### 1. When is the scope required to be completed?
no specified date 

  ### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
no dependencies

remingtontr, 2024-05-13:
Ready for initial review

amsiotdeveloper, 2024-05-19:
@remingtontr looks great!  Good work!  One small suggestion since UCI was not defined I thought we should mention it is the command-line utility first then in the next sentence we can refer to it directly as the UCI.  Should we actually define the acronym?  In this case I don't think we have to because it is the name of the utility? 

@greenbreakfast this article is ready for your review/approval.

remingtontr, 2024-05-19:
Yes. At the first instance of the word you use the full word followed by the acronym in brackets. After this you can use the acronym only.

Tom Remington

greenbreakfast, 2024-05-22:
@remingtontr looks good, just added one comment

greenbreakfast, 2024-05-28:
@remingtontr thanks for the update, please go ahead with a PR

closed 2024-05-31 by greenbreakfast