OnionIoT/documentation #83: Using WiFi & Ethernet Simultaneously

amsiotdeveloper, 2024-05-03:
**NOTE:** _This Subtask is linked to a larger Task List_ ticket #78 

# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development
        2. Deployment
        3. Production
      This article is primarily targeting the Development track to help users better understand the Omega's networking capabilities.
           
  ### 2. Create a short statement explaining the objectives and goals for this ticket.
The objectives of this article is to provide the reader with an understanding that both the WiFi and Ethernet can be used simultaneously in applications such as a router solution.

# **Description of the Scope:**

For the draft scope outline please refer to:

[https://docs.google.com/document/d/1w67dYSW5Kp9E4u7liLximAxnMgJMtOyQrj-JYH162yc/edit]

Anything in _italics_ is an instruction to the writer. The rest is content

This is a new article so it is more of a place holder and an FYI for the reader.  More technical content can be added in future when we circle back.

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 

See this article from old documentation: https://docs.onion.io/omega2-docs/fix-ip-addr-collisions.html

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 
- reuse content where possible
- create an article with a professional business/technical tone 
- create an article consistent with the style guide 

  ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.

working doc in google docs 
markdown pushed to repo 

# **Date Required By:**
  ### 1. When is the scope required to be completed?
no specified date 

  ### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
no dependencies

remingtontr, 2024-05-13:
Ready for initial review

amsiotdeveloper, 2024-05-19:
This is short article - looks good.
@greenbreakfast this article is ready for your review/approval.

greenbreakfast, 2024-05-23:
Thanks @remingtontr 
Please go ahead with a PR