OnionIoT/documentation #103: Special Pin - RST mod

amsiotdeveloper, 2024-06-04:
# **The Story for a New Chapter:**
### **Task Owner:**

Who owns this task and will approve the final deliverables? 
Lazar

### **The Scope:**
###  1. How does this task relate to the value track(s)?
- All of the effort we put into improving our product lines should provide value to our customers.
- We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
1. Development
2. Deployment
3. Production
This content is to help with development in the rare instances the user requires a hardware reset. 
        
###  2. Create a short statement explaining the objectives and goals for this Chapter.

The objectives of this chapter is to explain to users how to add the required modifications to get a hardware reset to reboot properly by correctly configuring the boot pins for a hardware reset.

- The goal is to summarize the blog post and rewrite it into our Special Pins article.
- include the table showing the boot pins from the blog
- include the schematic from the blog 
- utilize as much existing content as possible 
- Articles should be succinct, and minimalist.  

### 3. **Reference Material:**
- Provide hyperlinks for helpful and related information. 
https://community.onion.io/topic/3049/omega2-reboot-fix-official-solution

https://docs.google.com/document/d/16bs7giVRbnYx_-8ux8QyPblSn-Ihr4nQl4CUbBxIOYI/edit

-

greenbreakfast, 2024-06-04:
Will this be an addition to the Special Pins article or a new article?

amsiotdeveloper, 2024-06-05:
@greenbreakfast there is a section in the article that covers reset so I think it makes best sense to just add more details there about the hardware reset and how to do it.   Make a "how-to" instead of a "blog after thought"

remingtontr, 2024-06-07:
Ready for initial review. I have revised the existing article.

closed 2024-07-09 by amsiotdeveloper