OnionIoT/documentation #107: 1.3 Intro Chapter: What's New in V23.05.3

amsiotdeveloper, 2024-06-07:
**NOTE:** _This Subtask is linked to a larger Task List_ticket #97  

# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development - 
        2. Deployment
        3. Production
          This article targets 2 types of users:
 new users - need a simple process to learn how to get started without overcomplicating things. 
legacy users - need to learn what's new and what has changed so they know how to stay up to date 

  ### 2. Create a short statement explaining the objectives and goals for this ticket.

The goal of this article is to provide a high level overview of what's new in V23.05.3

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 
Please refer to outline: 
https://docs.google.com/document/d/1cu1a4PQIOfUiZmivo9sjIApu-k7bkSGaubqVQkJ7wt8/edit#heading=h.5za80p6hlxcd

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 

- Articles should be succinct, and minimalist.

 ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- a succinct article that follows our current style guide based on the original article

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no date has been specified 

 ### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- Ticket #97  has a dependency on this ticket to be completed

remingtontr, 2024-06-12:
Ready for initial review.

amsiotdeveloper, 2024-06-18:
@remingtontr I like how you reworked the bullet list.  Well done!  Just one suggestion to improve the tone on one of the sentences.

amsiotdeveloper, 2024-06-24:
@remingtontr good work - let's proceed with the PR

closed 2024-07-22 by amsiotdeveloper