OnionIoT/documentation #108: 1.4 Intro Chapter: Comparison of v0.3.4 & v23.05.3

amsiotdeveloper, 2024-06-07:
**NOTE:** _This Subtask is linked to a larger Task List_ticket #97  

# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development - 
        2. Deployment
        3. Production
          This article is primarily for legacy users so they can figure out more specific details about the migration and what they might need to address in their applications if they choose to use the new firmware. 

  ### 2. Create a short statement explaining the objectives and goals for this ticket.

The goal of this article is to educate legacy users about what has changed by providing a direct comparison between the legacy and new firmware images.

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 
Please refer to outline: 
https://docs.google.com/document/d/1cu1a4PQIOfUiZmivo9sjIApu-k7bkSGaubqVQkJ7wt8/edit#heading=h.5za80p6hlxcd

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 

- Articles should be succinct, and minimalist.

 ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- a succinct article that follows our current style guide based on the original article

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no date has been specified 

 ### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- Ticket #97  has a dependency on this ticket to be completed

remingtontr, 2024-06-13:
Ready for initial review

amsiotdeveloper, 2024-06-18:
Excellent work!  Just one question about the missing hyperlink?  I'm not sure if you intentionally left that link out but I would suggest keeping it as a reference so users know what we are talking about? 
I will follow your suggestions about this, so other than that this article is approved for production.

remingtontr, 2024-06-19:
I may have accidentally removed it or simply left it out for the article. I'll make sure it goes back in.

closed 2024-07-22 by amsiotdeveloper