OnionIoT/documentation #114: Omega 2

amsiotdeveloper, 2024-06-18:
**NOTE:** _This Subtask is linked to a larger Task List_ticket #113 

# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development - 
        2. Deployment
        3. Production
This article provides an overview of the Omega2 module which helps users decide which one is best suited for the development needs.   These are articles are targeting users early in the decision process when they are trying to figure out what module to develop with. 

  ### 2. Create a short statement explaining the objectives and goals for this ticket.

The objectives of the Omega2 article is to provide provide a comprehensive overview of the options available to a new user  and provide the with high level information they would use in their decision process.

- The goal is to utilize as much existing content as possible so we can progress quickly through these articles. 
- Articles should be succinct, and minimalist.  

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 
Please refer to outline: 
https://docs.google.com/document/d/1N0GZG83dOvpB2MxFTK3Ez3AGx8D9avBp6x9aHxjGCRg/edit

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 

- Articles should be succinct, and minimalist.

 ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- a succinct article that follows our current style guide based on the original article
- the article should provide enough detail for a new user to base a buying decision on

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no date has been specified 

 ### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- Ticket #113   has a dependency on this ticket to be completed

remingtontr, 2024-06-21:
Ready for initial review. I wasn't quite sure what you were looking for here, so I've included all diagrams from the Omega2 diagram PDF.

amsiotdeveloper, 2024-06-25:
@remingtontr I have quite a few suggestions/comments/markups for this article.  I'm sure a lot of this wording has been copied from the original documentation, but my biggest suggestion is to keep a professional tone.  The original docs were for the maker space, but now our audience is more in the B2B space, which is why I'm suggesting we clean up the tone.   Let's give this another revision please.

remingtontr, 2024-06-27:
I've revised the article for tone and made some edits.

amsiotdeveloper, 2024-07-02:
Much improved!  Good work.
I just found a couple small typo's - see doc for revisions notes. 
Please proceed with PR.

amsiotdeveloper, 2024-07-04:
@remingtontr agreed - let's remove that table, and proceed with the PR

greenbreakfast, 2024-10-15:
completed in pr #134