OnionIoT/documentation #115: 2.1.1 Omega2 Mechanical

amsiotdeveloper, 2024-06-18:
**NOTE:** _This Subtask is linked to a larger Task List_ticket #113 

# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development - 
        2. Deployment
        3. Production
This article provides an overview of the Omega2 module which helps users decide which one is best suited for the development needs.   These are articles are targeting users early in the decision process when they are trying to figure out what module to develop with. 

  ### 2. Create a short statement explaining the objectives and goals for this ticket.

The objectives of this article is to provide a mechanical drawing of the Omega 2 module for design consideration. 

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 
Please refer to outline: 
https://docs.google.com/document/d/1N0GZG83dOvpB2MxFTK3Ez3AGx8D9avBp6x9aHxjGCRg/edit

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 

- a succinct article with a mechanical drawing of the Omega 2 with dimensions 

 ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- a dimensioned drawing

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no date has been specified 

 ### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- Ticket #113   has a dependency on this ticket to be completed

remingtontr, 2024-06-21:
Ready for initial review

amsiotdeveloper, 2024-06-25:
@remingtontr looks great!  Please proceed with the PR

closed 2024-07-22 by amsiotdeveloper