OnionIoT/documentation #117: 2.2 Omega 2S

amsiotdeveloper, 2024-06-18:
**NOTE:** _This Subtask is linked to a larger Task List_ticket #113 

# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development - 
        2. Deployment
        3. Production
This article provides an overview of the Omega2S module which helps users decide which one is best suited for the development needs.   These are articles are targeting users early in the decision process when they are trying to figure out what module to develop with. 

  ### 2. Create a short statement explaining the objectives and goals for this ticket.

The objectives of the Omega2S article is to provide provide a comprehensive overview of the options available to a new user  and provide the with high level information they would use in their decision process.

- The goal is to utilize as much existing content as possible so we can progress quickly through these articles. 
- Articles should be succinct, and minimalist.  

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 
Please refer to outline: 
https://docs.google.com/document/d/1N0GZG83dOvpB2MxFTK3Ez3AGx8D9avBp6x9aHxjGCRg/edit

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 

- Articles should be succinct, and minimalist.

 ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- a succinct article that follows our current style guide based on the original article
- the article should provide enough detail for a new user to base a buying decision on

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no date has been specified 

 ### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- Ticket #113   has a dependency on this ticket to be completed

remingtontr, 2024-07-02:
Ready for initial review.

amsiotdeveloper, 2024-07-04:
Well done, I like the choice of the pic over the previous diagram.  Great idea!! 
Please proceed with PR.

greenbreakfast, 2024-10-15:
completed in pr #135

closed 2024-10-15 by greenbreakfast