OnionIoT/documentation #126: 3.2 Quick Start: Serial Command Line

amsiotdeveloper, 2024-06-26:
**NOTE:** _This Subtask is linked to a larger Task List_ticket #120   

# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development - 
        2. Deployment
        3. Production
This article helps users in the development stage so they can quickly learn how to setup and use the serial command line interface.

  ### 2. Create a short statement explaining the objectives and goals for this ticket.

The goal of this article is to:
- provide an introduction to the serial command line interface and the benifits 
- outline the steps required to setup the serial port 
- and connect the serial command line interface. 

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 
Please refer to outline: 
https://docs.google.com/document/d/1h7JF9ceYgBxK7cVass_cCJOiFn6moJTYrZp7dWuJpKI/edit#heading=h.wfyhhst8qxhk

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 

- Articles should be succinct, and minimalist.
- A step by step easy to follow outline. 

 ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- a succinct article that follows our current style guide based on the original article
- a new user should be able to easily follow the instructions to setup the serial port and command line interface

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no date has been specified 

 ### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- Ticket #120  has a dependency on this ticket to be completed

remingtontr, 2024-07-09:
> @remingtontr<https://github.com/remingtontr> feedback and suggestions for changes in the first section of the article. I think it flows better like this.
> Please review and revise accordingly.
> Thank you.

Agreed. Revised.

Tom Remington

amsiotdeveloper, 2024-07-10:
@remingtontr one small suggestion to tweak - other than that it looks good. Please proceed with PR.

greenbreakfast, 2024-10-15:
completed in pr #139

closed 2024-10-15 by greenbreakfast