OnionIoT/documentation #129: 3.4 Quick Start: Hello World Example

amsiotdeveloper, 2024-06-27:
**NOTE:** _This Subtask is linked to a larger Task List_ticket #120   

# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development - 
        2. Deployment
        3. Production
This article helps users in the development stage so they can quickly learn how to run a basic application and start developing on a known good piece of code.

  ### 2. Create a short statement explaining the objectives and goals for this ticket.

The goal of this article is to:

- give a new user a starting point to create a new application 
- provide working code for a basic application that they can use to start building a more advanced solution with 

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 
Please refer to outline: 
https://docs.google.com/document/d/1h7JF9ceYgBxK7cVass_cCJOiFn6moJTYrZp7dWuJpKI/edit

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 

- share links where to find the code for the basic Hello World demo application 
- outline the basic steps to show how to load and run the application 

 ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- a succinct article that follows our current style guide based on the original article
- a simple easy to follow outline to start using a basic application for development

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no date has been specified 

 ### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- Ticket #120  has a dependency on this ticket to be completed

remingtontr, 2024-07-15:
Ready for initial review

amsiotdeveloper, 2024-07-17:
@remingtontr  this is a good start.  Let's put this on hold until @greenbreakfast gets back and then he can support us with the required example code.

remingtontr, 2024-07-17:
Okay, sounds good.

Tom Remington

greenbreakfast, 2024-08-08:
@remingtontr added comments and hello world code to the outline

remingtontr, 2024-08-09:
I've revised the article based on your comments and added a background to the codeblock so it's easier to read in Google Docs.

greenbreakfast, 2024-08-14:
@remingtontr added a few more comments

greenbreakfast, 2024-08-15:
@remingtontr looks good, please go ahead with a PR

closed 2024-10-15 by greenbreakfast