OnionIoT/documentation #130: 3.5 Quick Start: Suggested Reading

amsiotdeveloper, 2024-06-27:
**NOTE:** _This Subtask is linked to a larger Task List_ticket #120   

# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development - 
        2. Deployment
        3. Production
This article helps users at all stages by providing links to suggested reading material.

  ### 2. Create a short statement explaining the objectives and goals for this ticket.

The goal of this article is to:

- start creating a list of suggested reading so a new user can refer to this article and easily find helpful content to get them started.

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 
Please refer to outline: 
https://docs.google.com/document/d/1h7JF9ceYgBxK7cVass_cCJOiFn6moJTYrZp7dWuJpKI/edit#heading=h.wfyhhst8qxhk

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 

- Articles should be succinct, and minimalist.
- an outline of links with a brief description why this is suggested content to read 

 ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- a succinct article that follows our current style guide based on the original article
- a new user should be able to find reading material to help the get started in different areas of their project development path 
- this article is intended to be a starting point and by no means do we expect at this stage to provide a comprehensive list.  We can build on this list as time permits. 

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no date has been specified 

 ### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- Ticket #120  has a dependency on this ticket to be completed

remingtontr, 2024-07-11:
For each of the products listed I'll provide a brief description. Do we want to link to the old documentation for a detailed reference or will we be rewriting the old documentation?

remingtontr, 2024-07-15:
Ready for initial review

amsiotdeveloper, 2024-07-15:
@remingtontr this looks like an excellent improvement from the original table.
My only suggestion is every item is starting out with "Onion's <item>..."
There is nothing wrong with this but after reading the whole doc it does get a little repetitive so if you think it's worth changing it up little then give it a shot, otherwise let's proceed with PR.

remingtontr, 2024-07-15:
Agreed. I'll revise slightly so it's not so repetitive.

greenbreakfast, 2024-10-15:
completed in pr #143

closed 2024-10-15 by greenbreakfast