OnionIoT/documentation #138: 6.2 Overview of Supported Languages

amsiotdeveloper, 2024-07-11:
# **Task Owner:**
Who owns this task and will approve the final deliverables? 

Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development
        2. Deployment
        3. Production
This article is to help the user understand the various languages available to develop code on the Omega platform.            

  ### 2. Create a short statement explaining the objectives and goals for this ticket.
The objective of this article is to provide an outline of the available languages so a user can have an easy reference to see what is avilable, and why they might want to use.  

    
# **Description of the Scope:**
  ### 1. Provide context while explaining the requirement of the ticket.
- create an outline of the supported languages
- provide a brief overview of each language
- explain why a user might want to use the language
- provide instructions how to compile and use the language 
- provide instructions on how to run the programs 

  ### 2. Provide sketches and diagrams where possible.

  ### 3. Add any other context or screenshots about the feature request here.

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 
https://docs.google.com/document/d/1vnaV-f_krxMhsoKz0G2WGIy1wKGXbEiWk734VFYLusM/edit

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 
- a clear outline of available languages 
- succinct 
- follow the style guide
-
  ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- a user should have a good understanding of the available languages, there should be enough information to help them make a decision on which language to use, and they should able to follow the instructions on how to compile and run programs for each language.

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no specified deadline
-
  ### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- no dependencies defined

remingtontr, 2024-07-18:
Ready for initial review

amsiotdeveloper, 2024-07-19:
@remingtontr this is excellent!  I think this provides good value!  Please proceed with PR.

greenbreakfast, 2024-10-15:
completed in pr #144

closed 2024-10-15 by greenbreakfast