OnionIoT/documentation #141: 8.1 How to build a firmware image

amsiotdeveloper, 2024-07-18:
# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development
        2. Deployment
        3. Production
          
This article is targeting the deployment and production value tracks. 

  ### 2. Create a short statement explaining the objectives and goals for this ticket.

The objective of this article is to provide users with an outline explaining how to build a custom firmware image and deploy it in production capacity. 
     
# **Description of the Scope:**
  ### 1. Provide context while explaining the requirement of the ticket.
- provide some context around custom firmware and the benefits 
- provide the method recommended by Onion 
- provide a step by step method for building a custom firmware image and deploying it
 
  ### 2. Provide sketches and diagrams where possible.
  ### 3. Add any other context or screenshots about the feature request here.

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 
https://docs.google.com/document/d/1xjcxFmpv22aukRAXqFi5GeabYcwKSx225I4CCz9RXX0/edit

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 
- provide instructional material that new users can understand and follow.

  ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- an easy to follow process for building and deploying customer firmware
- a new user should be able to follow the process and build a custom firmware image 

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no date specified

 ### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- no dependencies

greenbreakfast, 2024-08-14:
@remingtontr added comments and suggestions

greenbreakfast, 2024-08-19:
@remingtontr looks good, please go ahead with a PR

greenbreakfast, 2024-10-15:
completed in pr #152

closed 2024-10-15 by greenbreakfast