OnionIoT/documentation #142: 8.3 How Onion Firmware is Built

amsiotdeveloper, 2024-07-18:
# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - All of the effort we put into improving our product lines should provide value to our customers.
    - We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
        1. Development
        2. Deployment
        3. Production
          
This article is targeting the deployment and production value tracks. 

  ### 2. Create a short statement explaining the objectives and goals for this ticket.

The objective of this article is to provide users with an insight into how Onion Firmware is built. 
     
# **Description of the Scope:**
  ### 1. Provide context while explaining the requirement of the ticket.
- This is an existing article that outlines the process of how Onion internally builds firmware.
This is a new approach that makes use of a 2-step process. We explain the 2nd step in detail in articles 7.3 and 8.1 so that customers can do it themselves.
 
  ### 2. Provide sketches and diagrams where possible.
  ### 3. Add any other context or screenshots about the feature request here.

# **Reference Material:**
  ### 1. Provide hyperlinks for helpful and related information. 
https://docs.google.com/document/d/1xLXtvn9s5p4CQBhOw0RBZFxj9JL6njozMtF_4TTU30U/edit

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 
- provide an outline on how Onion Firmware is built
- make appropriate improvements on the existing article.  Improve the process flow, and language.
- make the article more succinct where possible
- article should be succinct and follow the style guide

  ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- an easy to follow process explaining how Onion Firmware is built.
- a new user should have a good understanding of the process

# **Date Required By:**
  ### 1. When is the scope required to be completed?
- no date specified

 ### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- no dependencies

remingtontr, 2024-07-25:
Ready for initial review. Please advise re comments in article.

greenbreakfast, 2024-08-14:
@remingtontr responded to comments, made comments/suggestions

greenbreakfast, 2024-08-19:
looks great @remingtontr !
Please go ahead with a PR

greenbreakfast, 2024-10-15:
completed in pr #153

closed 2024-10-15 by greenbreakfast