OnionIoT/documentation #147: Improve Packages chapter articles

greenbreakfast, 2024-08-14:
# **Task Owner:**
*Who owns this task and will approve the final deliverables? *
Lazar

# **Objectives & Goals:**
  ### 1. How does this task relate to the value track(s)?
    - *All of the effort we put into improving our product lines should provide value to our customers.*
    - *We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:*
        1. Development
        2. Deployment
        3. Production
           
All 3

  ### 2. Create a short statement explaining the objectives and goals for this ticket.
     
Improve upon the articles in Packages chapter

# **Description of the Scope:**

Articles and changes
* OpenWRT Packages 
  * reorganize some of this for clarity, cut the Firmware section to a brief para and link to the new Firmware article, maybe add just a brief para about firmware packages
  * this article is meant to define the key terms. consider renaming the article so it's clear we're just defining terms here
* OPKG Package Manager
  * rework so this goes from general to specific about Omega. Move the 2 advanced sections into their own article.
* Create a Custom Package
  * Move definitions to before the Makefile example
  * move Package Source Code section to before The Structure of the Package Source.
* Deploy a Package Repo
  * Move any content relating to compiling into the Compile a Package article and keep this article solely about deployment.
  * 
* Compile a Package
  * make sure we clearly separate and explain the difference between compiling a package that's in development, ie from local source code (content that's currently in the compilation article) and compiling a whole package feed for production (content that will be moved in from the Deploy a Package Repo article)
  * Add a note that the last step of compiling a whole package feed is creating an index of the packages and signing the packages. This is required for the compiled packages to be used as an opkg package repo by a device

# **Deliverables or Desired Outcome:**
  ### 1. What are the task owners expectations? 
  ### 2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.

Updates to existing articles

# **Date Required By:**
  ### 1. When is the scope required to be completed?
  ### 2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.

N/A

remingtontr, 2024-08-21:
All articles have been updated and are ready for initial review.

amsiotdeveloper, 2024-08-30:
@remingtontr  - as per the message I posted to you in slack: 

- [ ] clean up the googledrive chapter 7 so the articles have the same name as the website 

- [ ] cleanup the Omega Documentation Map spreadsheet so chapter 7 outline is the same as the website and links are to the correct articles in the googledrive 

After we get things back in order then we can review and merge the changes in each article:

- [ ] review and merge article on OpenWRT Packages 
- [ ] review and merge article on OPKG Package Manage
- [ ] review and merge article on Create a Custom Package
- [ ] review and merge article on Deploy a Package Repo
- [ ] review and merge article on Compile a Package

remingtontr, 2024-09-09:
The first part of this has been done.

I am currently waiting for the articles to be reviewed.

Tom Remington

remingtontr, 2024-10-03:
The first two sections in the Aug 30 comment are complete and the articles are waiting for review.

**Please note:** 

- It was greed that the OpenWRT Packages article should be renamed to better reflect the contents of the article. I have renamed the article Package Definitions.
- It was agreed that the advanced package topics would be moved to their own article. I called this article Advanced Package Management.

amsiotdeveloper, 2024-10-16:
@remingtontr Package Definitions (OpenWRT Packages) article looks good - ready for PR

amsiotdeveloper, 2024-10-17:
@remingtontr OPKG Package Manager is good to go - please send PR
https://docs.google.com/document/d/11rN47S1qF6waH3zp-fi0VJsyyZOQ_qvI/edit

amsiotdeveloper, 2024-10-17:
@remingtontr Create a Custom Package article is good to go - please send PR
https://docs.google.com/document/d/1hCWI6u3Fr7Od54hsXeDWCD7QrSPNKKcE/edit

amsiotdeveloper, 2024-10-17:
@remingtontr  Compile a package article is good to go - please send PR 
https://docs.google.com/document/d/1PFar96tas9gqdAlFrngEt0jkCR5o7g2q/edit

amsiotdeveloper, 2024-10-18:
@remingtontr the advanced package manager is ready to go to - please send PR
https://docs.google.com/document/d/15kMzOKswYwaEoS0Tf6sxcRwE4oWV6tZm/edit

remingtontr, 2024-10-18:
I created a PR yesterday for the updates to Compile a package, Create a custom package, OPKG package manager, and Package definitions. I think this might have already been deployed.

I added the Deploy a package repo and Advanced Package Management to the PR for Issue-147, but I'm not seeing the PR.

Everything is now done for Issue-147.

Cheers,

Tom Remington

https://docs.google.com/document/d/15kMzOKswYwaEoS0Tf6sxcRwE4oWV6tZm/edit

greenbreakfast, 2024-10-22:
completed in PRs #166, #167, and #168