OnionIoT/documentation #148: Update one-wire article to avoid ghost devices

greenbreakfast, 2024-08-15:
# What article does this bug/error affect?
*Provide either the URL or the path to the affected file*

One-Wire article in Hardware Interface chapter - https://documentation.onioniot.com/hardware-interfaces/one-wire

# **Describe the bug**
*A clear and concise description of what the bug/error or is.*

If the one-wire master pin on the Omega2 is not pulled down to ground or connected to a one-wire slave device after one-wire support is enabled, the system will detect "ghost" one-wire devices. These are devices that show up as one wire slaves but do not actually exist. 
This is likely caused by random fluctuations on the one-wire master pin.

# **To Reproduce**
Steps to reproduce the behavior:
1. Follow procedure from https://documentation.onioniot.com/hardware-interfaces/one-wire to enable one-wire support but do not connect any one-wire devices to GPIO2, instead leave it floating
2. Wait a few minutes
3. Check to see if any slave devices have been detected by running `cat /sys/devices/w1_bus_master1/w1_master_slave_count`
4. Observe a non-zero number of slave devices
5. Check the slave device unique IDs by running `cat /sys/devices/w1_bus_master1/w1_master_slaves`
6. Observe random slave device unique IDs

# **Expected behavior**
*A clear and concise description of what you expected to happen.*

Should not observe any slave devices in step 4,  list of slave device unique IDs should be empty in step 6

# **Screenshots**
*If applicable, add screenshots to help explain your problem.*

Example of ghost slave devices - the first 2 are ghost devices:
[image]

# Resolution

The article should be updated with a note to mention the one-wire master pin should not be left floating after one wire support is enabled. It should either be pulled to ground or connected to one or more one-wire devices.

remingtontr, 2024-08-22:
Ready for initial review.

amsiotdeveloper, 2024-08-23:
@remingtontr  this is the article I reviewed and ithe updates look good..
https://docs.google.com/document/d/1QSGYaw3LVpXHNzv39t9feIl00_WXkcyKMO577__R7oM/edit#heading=h.rgrg6hl5oqiy

- please move the doc to the correct folder and update the link in the map then proceed with the PR

greenbreakfast, 2024-10-15:
completed in pr #154

closed 2024-10-15 by greenbreakfast