OnionIoT/documentation #150: Add article on automatically running a command on system boot

greenbreakfast, 2024-08-19:
## Motivation
*Why are we doing this? What use cases does it support? What problems does it solve? What is the expected outcome?*

See user comment https://github.com/OnionIoT/documentation/discussions/146

### **Is your documentation request related to a problem? Please describe.**
*A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]*

It is useful to be able to configure the Omega to run commands automatically on boot. This can help reduce friction for users implementing their applications. 
Examples can range from setting gpio mux config to  

## **Describe the solution you'd like**
*A clear and concise description of what you want to happen.*

Add an article to the Software chapter.
Can be heavily based on the "Running a Command on Boot" article from the old documentation (article markdown source here: https://github.com/OnionIoT/Onion-Docs/blob/master/Omega2/Documentation/Doing-Stuff/Automation/Running-a-command-on-boot.md)

The article from the old documentation is fully example-based. For the new article let's first explain the concept(s) and then add example(s) as necessary to illustrate how to implement the concept.

Concepts:
- Using the `/etc/rc.local` file to run commands at boot
  - Commands can be individual commands or scripts can be executed
- The `/etc/rc.local` file must complete execution in order for the Omega to successfully complete the boot sequence
  - Scripts must end in `exit 0`
  - If the command that needs to be run at boot needs to run continuously, fork the process so it continues to run in the background `<YOUR COMMAND TO RUN> &`
    - This is generally not recommended - instead create a new service
- Saving the output of the `rc.local` commands (optional)
- Troubleshooting

Examples:
- For general `/etc/rc.local` use, let's use a gpio muxing command (see https://documentation.onioniot.com/hardware-interfaces/pin-multiplexing#changing-the-pin-function) since this is a real world use case
- For saving the output of the `rc.local` commands, can use `date` to save the boot time to a file in the `tmp` directory. Full example would be: `date >> /tmp/boot-date.txt 2>&1`

## **Describe alternatives you've considered**
*A clear and concise description of any alternative solutions or features you've considered.*

N/A

## **Additional context**
*Add any other context or screenshots about the feature request here.*

Feel free to follow up if the instructions for the article are too vague

remingtontr, 2024-08-22:
Ready for initial review.

amsiotdeveloper, 2024-08-30:
@remingtontr I have reviewed the article and we need @greenbreakfast to clarify one paragraph - see comments in article. 
I think we need @greenbreakfast  to clarify the section "A practical use case"
I added a comment in the article about tone so I will leave it up to the two of you to decide if it should be improved or not.

remingtontr, 2024-10-04:
I have revised the article based on the comments. Lazar has indicated the "A Practical Use Case" example is correct.

amsiotdeveloper, 2024-10-16:
@remingtontr looks good - please proceed with PR

closed 2024-10-17 by greenbreakfast