OnionIoT/documentation #151: Update GPIO mux article to include instructions on permanently setting gpio mux config

greenbreakfast, 2024-08-19:
## Motivation
*Why are we doing this? What use cases does it support? What problems does it solve? What is the expected outcome?*

See user comment https://github.com/OnionIoT/documentation/discussions/146

### **Is your documentation request related to a problem? Please describe.**
*A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]*

GPIO muxing commands are available but to meet the needs of user applications, it should also be explained how to make the configuration permanent

## **Describe the solution you'd like**
*A clear and concise description of what you want to happen.*

The most straight-forward way to make the gpio mux configuration permanent is to run the gpio mux configuration command at boot time automatically.

Let's add a section to the Pin Multiplexing article explaining that, to make the pin multiplexing config permanent, the omega can be configured to run the gpio mux command at boot time automatically. Link to the new "Running a command at boot automatically" article from https://github.com/OnionIoT/documentation/issues/150

## **Describe alternatives you've considered**
*A clear and concise description of any alternative solutions or features you've considered.*

Also possible to use a Device Tree Overlay to specify the gpio mux default behaviour. But this way is less work intensive and achieves the same result.

## **Additional context**
*Add any other context or screenshots about the feature request here.*

Depends on completion of https://github.com/OnionIoT/documentation/issues/150

remingtontr, 2024-08-22:
Ready for initial review.

amsiotdeveloper, 2024-08-23:
@remingtontr changes look good - please proceed with PR.
- also move doc to correct folder, and add a link to the map please.

amsiotdeveloper, 2024-08-23:
...and to be thorough, this is the doc I reviewed.
https://docs.google.com/document/d/1KE77l9Z-5c6Bpb2sabVxiKiF0HuVi6Rx9axM2pAw4BA/edit#heading=h.ivhpikjt7k46

remingtontr, 2024-08-23:
This is the correct document. I have added the word "updated" to the name so I don't over write the original document and moved it to the correct folder.

Tom Remington

https://docs.google.com/document/d/1KE77l9Z-5c6Bpb2sabVxiKiF0HuVi6Rx9axM2pAw4BA/edit#heading=h.ivhpikjt7k46

closed 2024-10-17 by greenbreakfast