OnionIoT/documentation #170: Update Compile a Package article structure for clarity

greenbreakfast, 2024-10-22:
# What article does this bug/error affect?

Compile a Package (https://documentation.onioniot.com/packages/compile-package/)

# **Describe the bug**

This article provides steps to compile packages that are in development (local package source code) and packages that are ready for production (package source code in a repo).

But:
1. The article doesn't directly articulate the difference between projects in development and production
2. The article contents are organized in a way that might cause confusion - procedure steps are interspersed with "educational" sections

# **Suggested Improvement**

To resolve the two issues listed above, I suggest updating the article to follow this heading structure:

- Packages in development vs packages in production
  - *brief comparison of the two*
  - What does it mean for a package to be in development
  - What does it mean for a package to be in production
    - What is a package feed?
- Method used to compile packages
- System setup
- Compiling packages for development
  - Config changes
  - compiling a package
- Compiling packages for production
  - Config changes
  - compiling a package feed

*Please note the heading names are suggestions meant to illustrate the structure. Feel free to change the text to something that flows better.*

Additionally, let's adjust the headers so that the table of contents doesn't include each individual step. Having all the steps in the table of contents makes the article feel intimidating in my opinion. 
See the [Build Custom Firmware article](https://documentation.onioniot.com/firmware/how-to-build-firmware) for reference. (Note: `####` headers don't show up in the TOC)

remingtontr, 2024-10-31:
I have revised the article as requested and updated the documentation map.

Ready for review.

greenbreakfast, 2024-11-01:
@remingtontr looks good, please go ahead with a PR

greenbreakfast, 2024-11-01:
@remingtontr looking good. I made a few suggestions in the google doc, please take a look

greenbreakfast, 2024-11-04:
Hi @remingtontr, thanks for the updates. Ready for a PR

closed 2024-11-05 by greenbreakfast