OnionIoT/documentation #191: Explain relationship of feeds and package source

greenbreakfast, 2025-01-29:
## Motivation
*Why are we doing this? What use cases does it support? What problems does it solve? What is the expected outcome?*

Feedback from @crispyoz 's [comment on Create a Custom Package article](https://github.com/OnionIoT/documentation/discussions/189#discussioncomment-11911492)

Want to better describe the package creation process, how package feeds play into the equation, and how to get started easily

### **Is your documentation request related to a problem? Please describe.**
*A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]*

More clarity is required on 1) how feeds work (what they are, how they hold package source), 2) how they relate to packages, and 3) how to quickly setup a feed so that you can compile your packages

> A section on setting up a feed would be useful. My sense is that new players just want to see their feed work, so a local feed is a simple way of making this happen, then reference(s) to the Openwrt docs on how to set up the different types of feeds. I mean the concept of any OpenWrt Makefile and it's "relationship" to the actual Makefile for a source feed is no intuitive.

## **Describe the solution you'd like**
*A clear and concise description of what you want to happen.*

Updates to Create a Custom Package article:
- Add section on feeds:
  - [ ] what is a feed and how does it relate to packages
  - [ ] what does a feed directory look like? Include a diagram, can link to example repo
  - [ ] how to initialize a feed? Link to compile a package article
- [ ] Specifically call out that package makefile is different from the actual application make file 

Updates to Compile a Package article:
- [ ] specifically call out that we recommend compiling packages using the OpenWRT SDK. Firmware images can be created with the OpenWRT Image Builder but this is a different tool. See the how onion builds fw article for more details
- In the "Compiling packages for development" section
  - [ ] specifically call out that the feed here is local
  - [ ] mention the setup step will update the package feeds
    - [ ] add this to "Compiling packages for production" section as well
- In the "Compiling packages for production" section
  - [ ] specifically call out that the feed here is a remote repo
  - [ ] mention the setup step will update the package feeds

## **Describe alternatives you've considered**
*A clear and concise description of any alternative solutions or features you've considered.*

N/A

## **Additional context**
*Add any other context or screenshots about the feature request here.*

N/A

greenbreakfast, 2025-02-10:
# Content for article additions

## Feed definition

Can reuse the package feed definition from the [Compile a package article](https://documentation.onioniot.com/packages/compile-package#what-does-it-mean-for-a-package-to-be-in-production)

### Feed directory structure

Can use this directory diagram
```
custom-feed/
|-- example-package-simple/
|   |-- files/
|   |   `-- config.txt
|   `-- Makefile
`-- example-package-complex/
    |-- files/
    |   `-- config.txt
    |-- patches/
    |   `-- 000-fix-typo.patch
    |-- src/
    |   |-- main.c
    |   `-- makefile
    `-- Makefile
```

remingtontr, 2025-02-14:
Lazar, I have made the requested changes to the file and uploaded a new version of the file on the Google Drive. Please review and let me know if further changes are needed.

Tomas